adafruit/Adafruit_BNO055 #3: No need to wait for Wire.available().

Koepel, 2015-05-10:
This line is not correct : "while (Wire.available() < len);".
It is not correct to wait for data after Wire.requestFrom(). When the Wire.requestFrom() returns, the I2C transmission has completely finished and the data is in the buffer.

tdicola, 2015-07-28:
Thanks for the tip, just tried it out and it works great without the wire available check.  Thanks again for suggesting it!

closed 2015-07-28 by tdicola

Koepel, 2015-07-28:
I have added this notifications to more code, just to be sure that it will updated when the code is revised. It's not a real "bug".

It can be improved even more (but that would be over-the-top):
The Wire.requestFrom() returns the number of received bytes.
That can be compared to the number of requested bytes. In the very rare case of a bad I2C bus or if a user has a multi-Master I2C bus with collisions, the number of received bytes could be zero.

commit f12bf4b5 (2015-07-28): Fix #3 and add calibration status to raw example.