adafruit/Adafruit_BNO055 #6: Consider adding UART mode support

tdicola, 2015-07-14:
This is a todo to add UART mode support to the library.  See section 4.7 of the datasheet for the details, it's just a simple I2C framing protocol over a serial connection.  This is really handy to have for devices that have problems with I2C clock stretching, like perhaps the ESP8266 (see https://github.com/adafruit/Adafruit_Sensor/issues/8#issuecomment-120691115 and https://github.com/esp8266/Arduino/issues/526).

tabchas, 2016-03-11:
@tdicola Any update on this?

ttww, 2016-03-11:
Maybe this is useful?
https://github.com/ttww/BNO055_AT_Commands_Orientation_Bluetooth_Server
But you need a small "converter" 'duino...

cyberluke, 2016-08-05:
I did this for great UART transport between two including correctly getting every packet: locate Arduino OSC packages and see SLIP serial example. With this u can send various structured data JSON-like over any two pins (RX/TX) between any two devices. If you have wireless, it can be wireless, of course.

Learn by watching code:
https://github.com/CNMAT/OSC/blob/master/examples/SerialSendMessage/SerialSendMessage.ino

Learn...old school:
http://cnmat.berkeley.edu/oscuino