adafruit/Adafruit_BNO055 #12: BNO055 Restore_Offsets Question

Navid360, 2015-12-07:
Hi
Calibration values are stored successfully using Restore_Offsets.
The only problem is: when I rerun the sensor (Power On) it asks for: "Move sensor slightly to calibrate magnetometers"
Does it mean that my sensor is not able to get_calibration values from EEPROM or it is a secondary magnetometer calibration?!

Thanks

zacwitte, 2016-02-12:
Looking at the code, it seems that means it did get the offsets from the EEPROM, but you need to move it slightly to finish the calibration.

Wetmelon, 2016-06-01:
I added that in because for some reason, the BNO055 will just sit and do nothing until you bump it.  Sometimes.  It doesn't enter Fusion mode, and I'm not sure why.  I changed the example a bit in https://github.com/Wetmelon/Adafruit_BNO055/commit/f5136614bc969b9cbd94985d5556367be164cf0b if you'd like to try it out without the "move sensor" thing.

microbuilder, 2018-01-11:
Closing since this is related to the IC design, not an issue with the repo itself. Feel free to re-open as a support issue on the forums for further discussion.

closed 2018-01-11 by microbuilder