adafruit/Adafruit_BNO055 #15: Active work on adding Interrupt Support

lessonpix, 2016-02-08:
Hi folks, great library (as always).  Is there any active work on adding basic interrupt support into this library?  From the datasheet, it looks like it's just a matter of writing the right code to the right registers via I2C.  I am looking to use this in a low-power project and I'd like to use the "Any Motion" interrupt to wake the MCU from a low-power state.

If not, I will take a stab at doing it myself, but I'd rather know if there's anyone on this since I can reorder my work to take advantage of it.

Thanks
Bill

ladyada, 2016-02-09:
No active work on it, will leave this iseue open but would appreciate any PR with interrupt code

tabchas, 2016-03-11:
Any updates on this? I might take a look on this myself.

microbuilder, 2017-01-15:
Closing this issue, though any PRs are always welcome!

closed 2017-01-15 by microbuilder