adafruit/Adafruit_BNO055 #29: wrong scaling for gyroscope

DrKipKnight, 2016-10-21:
BNO055 specifications state (4.3.60 UNIT_SEL 0x3B) default units for gyroscope is degrees per second.  Therefore the scale is wrong.  Physical test of rotation rate shows scale should be 16 as specified.

in function
imu::Vector<3> Adafruit_BNO055::getVector(adafruit_vector_type_t vector_type)

    case VECTOR_GYROSCOPE:
      /* 1rps = 900 LSB */
      xyz[0] = ((double)x)/900.0;
      xyz[1] = ((double)y)/900.0;
      xyz[2] = ((double)z)/900.0;

should be

    case VECTOR_GYROSCOPE:
      /* 1dps = 16 LSB */
      xyz[0] = ((double)x)/16.0;
      xyz[1] = ((double)y)/16.0;
      xyz[2] = ((double)z)/16.0;

microbuilder, 2017-01-15:
Sorry for the (extremely) slow reaction time on this, but you seem to be right looking at the DS. I originally set the driver to output rps but then later commented that out, so it now uses the default dps value.  I need to do some tests on this before I can commit a change and I don't have a BNO055 in front of me, but I'll get a fix out for this shortly.

DrKipKnight, 2017-01-15:
I believe the Python code is the same.
Kip

microbuilder, 2017-04-20:
Fixed in https://github.com/adafruit/Adafruit_BNO055/commit/2b220edcf979f8996dcd7ec2e8d8252bde31ee66

closed 2017-04-20 by microbuilder