adafruit/Adafruit_BNO055 #33: Support of motion detection

show0k, 2017-01-18:
Accelerometer and gyroscope "any motion" and "no motion" events triggered by an interrupt pin (p39-45 & p53 of the [BN055 documentation ](https://cdn-shop.adafruit.com/datasheets/BST_BNO055_DS000_12.pdf)) are not supported by this library.

These are very interesting features for any project which has other thing to process than 100Hz of acc/gyr samples.

mrkaroshi, 2017-03-06:
Implementing this currently for a research project I'm working on, however I'm not sure how best to format the required functions; I could write one function such as 'enableMotionInt' to specify which type of interrupt should be linked to the INT pin but that might be a bit reductive if you want to enable several different types then simply call the status afterwards to see which triggered.

mrkaroshi, 2017-03-06:
Alternatively I could add a series of macros to the header allowing the chaining of several different interrupt functions, all of which would be automatically linked to the pin. To make use of either of these functions properly the developer would need a second function to link their embedded device's interrupt pin to the BNO055, but that may restrict anyone wanting to use C to handle their interrupts instead.

mrkaroshi, 2017-03-21:
Added a series of functions in my pull request, check my fork for interrupt functions.

show0k, 2017-03-21:
Thank you, I will try to use it next week !

microbuilder, 2018-01-11:
Related to #39 so closing this issue to avoid duplication.

closed 2018-01-11 by microbuilder