adafruit/Adafruit_BNO055 #35: setSensorOffsets(const uint8_t* ...) writes to wrong registers

marc22alain, 2017-02-06:
- Arduino board:  found on Teensy 3.1, but would affect all micro-controllers

- Arduino IDE version (found in Arduino -> About Arduino menu):  found with v1.8.1, but would affect all versions

The following operation should simply save and restore the offsets:

```
uint8_t savedCalibration[22];
bno.getSensorOffsets(savedCalibration);
bno.setSensorOffsets(savedCalibration);
```

However, the method `void  setSensorOffsets(const uint8_t* calibrate);` has switched the registers that store the calibration offsets for the magnetometer and gyro. The sensor offset register map shown on page 51 of the BNO055 specification sheet shows the order of registers as: accelerometer (55 - 5A), magnetometer (5B - 60), gyro (61 - 66), acc radius (67 - 68), mag radius (69 - 6A).

The actual order used in the method is: accelerometer, gyro, magnetometer, acc radius, mag radius. 

```
    write8(ACCEL_OFFSET_X_LSB_ADDR, calibData[0]);
    write8(ACCEL_OFFSET_X_MSB_ADDR, calibData[1]);
    write8(ACCEL_OFFSET_Y_LSB_ADDR, calibData[2]);
    write8(ACCEL_OFFSET_Y_MSB_ADDR, calibData[3]);
    write8(ACCEL_OFFSET_Z_LSB_ADDR, calibData[4]);
    write8(ACCEL_OFFSET_Z_MSB_ADDR, calibData[5]);

    write8(GYRO_OFFSET_X_LSB_ADDR, calibData[6]);
    write8(GYRO_OFFSET_X_MSB_ADDR, calibData[7]);
    write8(GYRO_OFFSET_Y_LSB_ADDR, calibData[8]);
    write8(GYRO_OFFSET_Y_MSB_ADDR, calibData[9]);
    write8(GYRO_OFFSET_Z_LSB_ADDR, calibData[10]);
    write8(GYRO_OFFSET_Z_MSB_ADDR, calibData[11]);

    write8(MAG_OFFSET_X_LSB_ADDR, calibData[12]);
    write8(MAG_OFFSET_X_MSB_ADDR, calibData[13]);
    write8(MAG_OFFSET_Y_LSB_ADDR, calibData[14]);
    write8(MAG_OFFSET_Y_MSB_ADDR, calibData[15]);
    write8(MAG_OFFSET_Z_LSB_ADDR, calibData[16]);
    write8(MAG_OFFSET_Z_MSB_ADDR, calibData[17]);

    write8(ACCEL_RADIUS_LSB_ADDR, calibData[18]);
    write8(ACCEL_RADIUS_MSB_ADDR, calibData[19]);

    write8(MAG_RADIUS_LSB_ADDR, calibData[20]);
    write8(MAG_RADIUS_MSB_ADDR, calibData[21]);
```

Wetmelon, 2017-02-24:
That explains why this never worked quite right lol

microbuilder, 2018-01-11:
This should be fixed in https://github.com/adafruit/Adafruit_BNO055/commit/5565ed3497994fc74c18e9270ff74e205e8c839b

closed 2018-01-11 by microbuilder