adafruit/Adafruit_BNO055 #44: The andafruit_bno055_offsets_t should be signed and not unsigned !

pmancele, 2017-06-05:
Correct me if i'm wrong but i think that the values you get from any sensor offet registers (LSB +MSB) are representing a signed value and not unsigned. So it should be `int16_t` and not `uint16_t` in the `adafruit_bno055_offsets_t`
```c++
 typedef struct
{
    uint16_t accel_offset_x;
    uint16_t accel_offset_y;
    uint16_t accel_offset_z;
    uint16_t gyro_offset_x;
    uint16_t gyro_offset_y;
    uint16_t gyro_offset_z;
    uint16_t mag_offset_x;
    uint16_t mag_offset_y;
    uint16_t mag_offset_z;

    uint16_t accel_radius;
    uint16_t mag_radius;
} adafruit_bno055_offsets_t;
```
Looking at the BNO055 datasheet it is mentionned that values can be negatives !

ethangrinnell78, 2018-01-08:
I agree with pmancele, they should all be **signed** 16-bit integers. The documentation states that these are +/- offset values. If we interpret this as an unsigned int then we can't have negative values. Changing them to int16_t corrected my offsets to numbers that made sense instead of huge values that were out of the given range.

In the data sheet see tables 3-16, 3-21, and 3-24 for accel, gyro, and radius offset ranges. Also, the paragraph in the middle of page 32 lists the magnetometer offset ranges, its not provided in a tabular format since it doesn't vary.

microbuilder, 2018-01-11:
Thanks for filing this, and sorry it slipped under the radar. An update (1.1.4) was just pushed out changing the values to signed.

closed 2018-01-11 by microbuilder