adafruit/Adafruit_BNO055 #48: Changing the default configuration setting of BNO055

mrrobles09, 2018-03-12:
As I was scrolling on the data sheet of BNO055 I notice that the Acc_config, Gyro_config 0, Gyro_config 1, and Mag_config were all in "read only".  If this is in read only then how will the user put their desire to set up when they need to? Also, will fusion mode overwrite the inputted set up when it is called at the end of the setup routine? See the example problem application below:

[image]

[image]

[image]

[image]

(The yellow highlighted part were related to the said questions)

Reference link:  www.mouser.com/ds/2/783/BST_BNO055_DS000_13-838248.pdf . Page 53, Register map Page 1.

Tones29, 2018-07-23:
If I understand p. 20-22 of the latest manual ((https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST_BNO055_DS000_14.pdf) correctly, the workflow is as follows:
 1) The sensor starts in `CONFIGMODE`. "This is the only mode in which all the writable register map entries can be changed. (Exeptions from this rule are the interrupt registers (INT and INT_MSK) and the operation mode register (OPR_MODE), which can be modified in any operation mode.)" [p.22]
 2) The user does the configuration. According to the manual, the configuration registered mentioned above are r+w [p.78, p.79], so I guess that just the color in the table is wrong.
 3) After setting `OPR_MODE` to anything else than `CONFIGMODE`, the configuration values cannot be changed any more. Setting a fusion mode should not change these values.

anziboy, 2019-05-20:
Hi, Im trying to change my OPR_MODE to IMU only, since I want to suspend the magnetometer, I'm wondering if you figured it out how to do it? I tried to change my OPR_MODE settings in my Adafruit library, but did not succeed. The sensors just turns off and shows no values.

Any tips? 
thank you

caternuson, 2021-08-12:
I think this was just errata. Looks fixed in current datasheet.
[image]

closed 2021-08-12 by caternuson