adafruit/Adafruit_BNO055 #50: Gyroscope output is in deg/s

shen-tian, 2018-05-21:
Not a problem per say, but looks like the library outputs gyro data in degrees/s, whereas [the tutorial](https://learn.adafruit.com/adafruit-bno055-absolute-orientation-sensor/arduino-code) claims it's in radians per second?

Reasons for believing this to be so:

1. When moving the sensor around, the output easily reaches >100, which is around 1000rpm had it been in r/s.

2. Reading the source: I see the unit setting code in the lib is commented out:

https://github.com/adafruit/Adafruit_BNO055/blob/e02f9bd4bcf9178d7f25194feb594d1ab9711435/Adafruit_BNO055.cpp#L93-L100

and this code

https://github.com/adafruit/Adafruit_BNO055/blob/e02f9bd4bcf9178d7f25194feb594d1ab9711435/Adafruit_BNO055.cpp#L343-L347

divides the raw output by 16, which is right for degree output (for radian output, it would have been to divide by 900).

Mainly a documentation issue (though I know the Unified Sensor lib prefers r/s, but that would be a breaking behaviour for this lib...)

- Feather M0
- PlatformIO v5.3.2
- Adafruit BNO055 v1.1.6

kevinjwalters, 2020-03-30:
As noted, Unified sensor docs also refer to "rad/s" in https://learn.adafruit.com/using-the-adafruit-unified-sensor-driver/how-does-it-work#standardised-si-units-for-sensor-data-2-2

caternuson, 2021-08-13:
Fixed with #110

closed 2021-08-13 by caternuson