adafruit/Adafruit_BNO055 #63: Sensor not found on ESP32

benkuper, 2019-02-04:
Hello,
i'm using the BNO-055 with an ESP32 (custom board, no breakout)
The BNO-055 seems to pass the I2C check but I can't connect with the Adafruit library to it.
It may not be a problem from the library as the devices is not scanned as well, but maybe people here have solutions for that still !
I came accross this issue on the esp32-arduino git : https://github.com/espressif/arduino-esp32/issues/81

I'm using pins SDA 2 / SCL 4 and BNO is on address 0x28

Thanks !

cyberluke, 2019-02-04:
Try to run i2c scanner example for arduino. It must find it.

cyberluke, 2019-02-04:
I have ESP32 and BNO055 at home. I was playing with older ESP. But I would like to help you and test it with you. Currently I'm working with NRF52, because they have much better Bluetooth configuration options.

Anyway, on every board, you need to setup I2C pins with pullup resistor. Either internal or put some resistor there. For interrupt pin, you need pulldown resistor.

benkuper, 2019-02-04:
Thanks, so I had people test other boards and it seems to be actually fine, I may have a faulty board. i'll be able to test soon other boards myself and confirm that.
One thing that could be improved in the library though is that the call to bno.begin() will force a call to Wire.begin() with no argument (inside begin() function), so if I want to override the sda/scl pins used with Wire, I have to modify this.
Maybe adding an alternative begin() function with sda/scl pins there would be nice !

cyberluke, 2019-02-04:
You mean you might have faulty ESP32 board? If you can upload hello world app and see text output, it should be ok. Maybe you need update ESP32 firmware. Let me know about your progress, please.

ladyada, 2019-03-07:
hiya closing because this is resolved! You can pass in any Wire object, your board definition should have the pins set up for that Wire device - that's the most flexible way to manage custom i2c setups

closed 2019-03-07 by ladyada