adafruit/Adafruit_BNO055 #65: Master branch is not working with the example code on Adafruit website

sinhau, 2019-02-21:
The example code on this [link](https://learn.adafruit.com/adafruit-bno055-absolute-orientation-sensor/arduino-code) doesn't seem to work with the master branch of this library.  The code seems to get hung up on bno.begin() line.  Switching the library to the 1.1.6 release fixes the issue.  

The other sample sketches such as rawdata.ino or sensorapi.ino in the examples folder would also brick my board soon after it uploads.  I would need to reflash it with the bootloader to get the board to work again.  This issue also got resolved when I reverted back to the 1.1.6 release.

I'm using a custom PCB with Arduino Micro and the BNO055 using an external crystal.

ladyada, 2019-02-24:
we've added documentation lately 

@hoffmannjan please take a look at what could have broken in the doxygen additions. did you test the board after the merge?

hoffmannjan, 2019-02-25:
@sinhau @ladyada it's fixed now, the problem was missing assigment to `_wire`, didn't tested after merge but that also why the version wasn't bumped :).
Tested today with all examples and all works fine.

closed 2019-02-25 by hoffmannjan