adafruit/Adafruit_BNO055 #67: getEvent returns incorrect type for linear acceleration data

potto216, 2019-04-17:
When requesting linear acceleration data using `getEvent` The type returned is  `SENSOR_TYPE_ACCELEROMETER` instead of `SENSOR_TYPE_LINEAR_ACCELERATION` .
This occurs when running the code under:
- Arduino board:  UNO/R3
- Arduino IDE version 1.8.9

To reproduce the problem create an IMU object and call `getEvent` requesting linear acceleration:
````C++
  Adafruit_BNO055 bno = Adafruit_BNO055(55);
  sensors_event_t linearAccelData;
  bno.getEvent(&linearAccelData, Adafruit_BNO055::VECTOR_LINEARACCEL);
````
The returned `linearAccelData.type` is equal to `SENSOR_TYPE_ACCELEROMETER` instead of `SENSOR_TYPE_LINEAR_ACCELERATION`.

This appears to be due to an assignment typo Adafruit_BNO055.cpp
````C++
bool Adafruit_BNO055::getEvent(sensors_event_t *event, adafruit_vector_type_t vec_type)
{
  ...
  if (vec_type == Adafruit_BNO055::VECTOR_LINEARACCEL)
  {
    event->type = SENSOR_TYPE_ACCELEROMETER;
    vec = getVector(Adafruit_BNO055::VECTOR_LINEARACCEL);

    event->acceleration.x = vec.x();
    event->acceleration.y = vec.y();
    event->acceleration.z = vec.z();
  } 
...
}
````

which should be:
````C++
    event->type = SENSOR_TYPE_LINEAR_ACCELERATION;
````
If this is an error I can create a pull request.  Thanks, Paul

ladyada, 2019-04-17:
yah please PR :)

potto216, 2019-04-17:
I just submitted the pull request which is my first public pull request. Let me know of any issues I should correct.

ladyada, 2019-04-17:
you did great!

PR #68 (2019-05-15): Corrected a typo that caused getEvent to return the incorrect type when using linear acceleration data. 
This pull request fixes #67. To correct a typo that caused getEvent to return the incorrect type when using linear acceleration data. I modified Adafruit_BNO055.cpp and changed a single line to have the linear acceleration data type set to  `event->type = SENSOR_TYPE_LINEAR_ACCELERATION`.

This change should be supported on all platforms and should not introduce any limitations.

I tested the code by running the change on an UNO/R3 and verifying the `SENSOR_TYPE_LINEAR_ACCELERATION` is retu [... cut]
closed 2019-05-15 by hoffmannjan