adafruit/Adafruit_BNO055 #69: BNO055 Euler Heading

BPiag, 2019-04-29:
Dear All,
I am working on a marine application of model surface vessel in which the 9-axis BNO055 is useful for attitude determination and eventual linear acc evaluation (to be later on completed with GPS Kalman Filtering). I have some doubts not cleared completely :

1. Is the default mode the NDOF?
2. In the default mode are the Euler Angles referred to north?
(in my case did not, neither after some time.. leading to the 1st doubt)
3. Does anybody observe drift in the Euler Angle heading (supposing it rotated to refer to nord across magnetometers at initial time?)?

In my case Euler heading drifts a lot in time: hence again leading to the doubt 1.. Shouldn't the fusion process unbias measurments across Mag in combination with GYRO? In my tests the Euler heading result much much more similar to a trapeze integration than a fusion.

In figure below:
Mag Heading in blue,
Euler heading (manually initialized to Mag heading) in orange,
external GPS course over ground in black (confirming MAG)
brutal trapeze integration of gyro rate (very similar to Euler) in red

Some sailing and turning

I tested two different IMU on two different vessels, leading to the same qualitative conclusions.
The calibration of GYRO and ACC is executed in lab and stored in order to be reloaded, while MAG are calibrated all start up. Finally: I read about the euler mismatch with quaternions of the BNO055 library, but just when tilting the IMU. Does it affect also heading fusion?
Regards

[img width=500 height=112]https://varuj77995.i.lithium.com/t5/image/serverpage/image-id/2592iDAE3AB549656E3F7/image-size/large?v=1.0&px=999[/img]
[https://varuj77995.i.lithium.com/t5/image/serverpage/image-id/2592iDAE3AB549656E3F7/image-size/large?v=1.0&px=999](url)

My setup: Arduino Mega 2560
Libraries:
#include <Adafruit_Sensor.h>
#include <Adafruit_BNO055.h>
#include <utility/imumaths.h>

pvonmoradi, 2021-01-25:
So you are going to re-calibrate MAG (which is a non-trivial procedure, like orthogonal 8 sign) every time you turn on your device?

caternuson, 2021-08-12:
The default mode is NDOF:
https://github.com/adafruit/Adafruit_BNO055/blob/c70a80e3ce57294cb89990b150982a4d60929159/Adafruit_BNO055.h#L286
https://adafruit.github.io/Adafruit_BNO055/html/class_adafruit___b_n_o055.html#aeea25ebb2984c57ffd1292e87ba336cb

For the rest of this, please consult the datasheet and other Bosch documentation on the sensor.

closed 2021-08-12 by caternuson