adafruit/Adafruit_BNO055 #70: Not compatible with other wire libraries

Dealerpriest, 2019-06-25:
I'm not able to use this library together with the teensy specific wire library ( `#include <i2c_t3.h>`)

I would request that the bno library doesn't include wire.h inside its code. Then one could provide ones own wire implementation.

ladyada, 2019-06-25:
hiya - we would not be able to do it - too difficult to support  - you can fork and adapt if you like!

closed 2019-06-25 by ladyada

Humancell, 2019-12-12:
I'm curious about this.  It appears that the constructor allow for a TwoWire compatible object to be passed as the third argument.  Wouldn't this allow you to use SoftwareWire with this library?

If this feature was coded in here, why is it too hard to support?

Humancell, 2019-12-12:
Actually ... I completely forgot!  I already hacked a version of this library to support SoftwareWire!

https://github.com/Wovyn/Adafruit_BNO055