adafruit/Adafruit_BNO055 #81: Accelerometer bandwidth

CherryGoose, 2020-05-16:
Hello! Im currently trying to set up my bno055 to estimate its position and im looking for a way to change accelerometer bandwidth from 62.5 Hz to something like 100 Hz or 200 Hz for more accurate measurement. Can you tell me how exactly i can do that? Im interested if one can change bandwidth to later use it in fusion mode. 
P.S. Im using arduino nano to communicate with bno055.

ladyada, 2020-05-17:
it looks like bandwidth is auto-controlled in fusion mode
[image]

caternuson, 2021-08-12:
Closing this in favor of #110. Can add general sensor config change requests there.

closed 2021-08-12 by caternuson