adafruit/Adafruit_BNO055 #84: Remove _wire->begin()

kiranshila, 2020-07-14:
Calling begin on a wire object in a library is bad practice as it resets the bus.

Doing so makes it imposible to use another `TwoWire` object, like one assigned for different SERCOM pins on SAMD boards as they require `pinPeripheral` to be set after `begin`, and calling `begin` again in the library resets this.

ladyada, 2020-07-14:
hi, this is not "bad" practice, it is the standard practice for all Arduino libraries including official ones
https://github.com/arduino-libraries/Arduino_LSM6DS3/blob/master/src/LSM6DS3.cpp#L76

 if you have reproducible code you can open an issue against the SAMD repo, begin() should not change the pin allocation or usage.
thanx!

closed 2020-07-14 by ladyada

kiranshila, 2020-07-14:
> begin() should not change the pin allocation or usage.

In fact it does https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/Wire/Wire.cpp#L37

ladyada, 2020-07-14:
great, you can open an issue there

kiranshila, 2020-07-14:
I'm just following your guide here https://learn.adafruit.com/using-atsamd21-sercom-to-add-more-spi-i2c-serial-ports/creating-a-new-wire
 Which clearly states that we have to call pinPeripheral immediately following `begin`. Does that then imply that this method of creating a new `TwoWire` object just doesn't work when used with libraries? If so, that's a pretty big caveat that should be mentioned.

kiranshila, 2020-07-14:
> great, you can open an issue there

Is that actually the issue though?

ladyada, 2020-07-14:
yes

kiranshila, 2020-07-14:
Done