adafruit/Adafruit_BNO055 #96: Roll Pitch and Headings

DrJaymz, 2020-11-16:
I am getting sensible data but not reporting via the properties I'm asking.  My heading is coming from orientation.roll and roll is coming from orientation.heading.  Pitch looks ok.

Just in case I thought I was going mad I can move it around to get the limits of the outputs which allows you to see clearly which is which.

So roll gives me a value between 0 and 360 which isn't valid for roll but is for heading.
Pitch is right because it gives me +/- 90 which is sensible because exceeding 90 by 10 degrees is the same as 80 degrees but reciprocal heading.  Heading gives me +/- 180 - that can be valid for heading but is more likely roll where -180 and 180 are the same - and it wouldn't really make sense as a heading because its supposed to be an absolute figure.

I could check these out in the web visualiser but that doesn't want to run on any of my computers or os'.

ladyada, 2020-11-16:
the absolute orientation may need to be rotated based on your usage. also you should use quaternions not euler to avoid errors at 90/180/270

closed 2020-11-16 by ladyada

DrJaymz, 2020-11-16:
I rotated the axis which is the correct way to do it according to the datasheet.  If you leave the axes as are and then apply rotations to the orientation then you won't get the right answers because the fusion algorithm uses magnetometer differently depending on the tilt with respect to its z axis.
  
I thought the answer may be due to using Android orientation, but that doesn't really make sense.

pvonmoradi, 2021-01-25:
@DrJaymz In my case, roll is around -Z, pitch is around -Y and yaw is around -X. So roll and yaw are switched and have the mirrored sign. Yaw is correct but its sign is incorrect.

MZandtheRaspberryPi, 2023-09-17:
I think there is a bug in that for euler angles the z rotation is read from the sensor and put into the x part of the output vector. And the x rotation from the sensor gets put into the z part of the vector. There is also some weirdness with signs, but i didn't look into that.

I think this is because the library reads from the registers in this order [here](https://github.com/adafruit/Adafruit_BNO055/blob/da2ea3b549d6035c2a800259f1d601b63bef84e6/Adafruit_BNO055.h#L122), which is heading, roll, pitch. It then writes heading to X, roll to Y, and pitch to Z [here](https://github.com/adafruit/Adafruit_BNO055/blob/da2ea3b549d6035c2a800259f1d601b63bef84e6/Adafruit_BNO055.cpp#L410C1-L414C58).

I've seen conflicting info out there on whether pitch corresponds to x rotation, or pitch corresponds to y rotation. I've tested with this sensor and this breakout board and the data in pitch register corresponds to X rotation in default axis mapping.
```
  BNO055_EULER_H_LSB_ADDR = 0X1A,
  BNO055_EULER_H_MSB_ADDR = 0X1B,
  BNO055_EULER_R_LSB_ADDR = 0X1C,
  BNO055_EULER_R_MSB_ADDR = 0X1D,
  BNO055_EULER_P_LSB_ADDR = 0X1E,
  BNO055_EULER_P_MSB_ADDR = 0X1F,
```

```
  readLen((adafruit_bno055_reg_t)vector_type, buffer, 6);

  x = ((int16_t)buffer[0]) | (((int16_t)buffer[1]) << 8);
  y = ((int16_t)buffer[2]) | (((int16_t)buffer[3]) << 8);
  z = ((int16_t)buffer[4]) | (((int16_t)buffer[5]) << 8);
```