adafruit/Adafruit_BNO055 #100: Wrong license? Shouldn't this library be GPL v3?

walchko, 2021-01-18:
Hello, I was poking around and noticed the license on the vector, quaternion, etc files you have in the `utility` directory has this:

```
    Inertial Measurement Unit Maths Library
    Copyright (C) 2013-2014  Samuel Cowen
        www.camelsoftware.com
    Bug fixes and cleanups by Gé Vissers (gvissers@gmail.com)
    This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.
    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.
    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
```

This is GPL code ... how is this library then MIT? I would suggest ditching this code, I don't think it is really fully utilized any ways and you could just use a simple `struct` (or something) to hold the data ... JMHO

ladyada, 2021-01-18:
thanks, if you can submit a PR for the swapped-in version that is freely licensable that would be awesome!

walchko, 2021-01-18:
ok

walchko, 2021-01-18:
#101 ok ... how is this?

walchko, 2021-01-18:
oh, should I have bumped the version number in the `library.properties` file?

supercamel, 2021-09-12:
Hello, I'm the original author of that code. It's been floating around the internet in various forms for years and I have no problem changing it to MIT or similar license.

ladyada, 2021-09-12:
@supercamel ok would you be ok publishing the new licensed code somewhere? we'll swap it in. thank you!!!

supercamel, 2021-09-23:
Sure, here it is

https://github.com/supercamel/imumaths

ladyada, 2021-09-23:
@caternuson at your convenience!

caternuson, 2021-09-24:
Done with #111 

Huge thanks to @supercamel for supporting this! :heart:

closed 2021-09-24 by caternuson