adafruit/Adafruit_BNO055 #116: Add Support for RP2040 Chip (Raspberry Pi Pico)

TristanMcGinnis, 2022-10-06:
I've tested a multitude of other Adafruit i2c devices with the new Pi Pico board and have had no issues thus far with compilation, sensor reading, etc... However, the BNO055 address is not read when using the RP2040 (Pi Pico/ KB2040). There is no issue uploading code, with Arduino IDE, however the sensor is not found by the microcontroller.

I seriously believe there is an issue with the library causing this, as the Adafruit unified sensor library, bmp3xx, and LSMDSOX, for example have all worked fine. The BNO055 works perfectly well with circuit python and its accompanying library(s), but not with Arduino.

TristanMcGinnis, 2022-10-26:
Bump. Any updates?

ladyada, 2022-11-10:
you can try
https://github.com/adafruit/Adafruit_BNO055/tree/timingtweaks
RP2040 core doesnt do clock stretching nicely it seems (?) but changing the freq makes it work on qtpy rp2040 with philhower core for us!

caternuson, 2022-11-14:
@TristanMcGinnis Please try the 1.6.1 release when it becomes available in Library Manager:
https://github.com/adafruit/Adafruit_BNO055/releases/tag/1.6.1

caternuson, 2022-11-23:
Closing due to lack of response. Release linked above includes update which should fix issue.

closed 2022-11-23 by caternuson